LazeMSS/OctoPrint-UICustomizer #283: Unresponsive UI in Firefox

MrVenerabilis, 2023-03-16:
This plugin looks nice so far.

When the UI is opened in desktop Firefox, the webcam stream wont load and the UI becomes unresponsive.
When changing the tab, the whole tab crashes and a loading symbol is displayed.
The UI and the plugin are working perfect in an other browser or on mobile firefox.

I tried to disable all plugins, but still not working.

**Steps to reproduce the behavior:**
1. Install the plugin
2. changes settings or not, doesnt matter.
3. see unresponsive UI

**Desktop:**
 - OS: Windows 11
 - lastest firefox

LazeMSS, 2023-03-19:
This is expected to be fix in the upcoming 0.1.9.4 release combined with the OctoPrint 1.9.0rc3 and later versions.

LazeMSS, 2023-03-22:
This should now be fixed when 1.9.0 is released - closing for now - wait for 1.9.0 release or try the RC3 version.
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.4

closed 2023-03-22 by LazeMSS

MrVenerabilis, 2023-03-22:
Perfect, thank you very much! Will try it out soon