LazeMSS/OctoPrint-UICustomizer #284: Octolapse recording text [BUG]

Awashcard0, 2023-03-22:
## Description
**Octolapse recording text gets moved
[image]
Describe your problem?
No problem just thought I should put it out there
Can you recreate it? 
[image]
Did you try disabling plugins?
No problem
Did you remember to update the subject?**
<hr>

**Plugins installed**
- ABL Expert Plugin[ABL_Expert] v0.6
- Creality Temperature[CrealityTemperature] v1.2.4
- Creality-2x-temperature-reporting-fix[ender3v2tempfix] v0.0.4
- Dashboard[dashboard] v1.19.9
- Detailed Progress[detailedprogress] v0.2.7
- Firmware Updater[firmwareupdater] v1.14.0
- Octolapse[octolapse] v0.4.1
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.0
- **UI Customizer[uicustomizer] v0.1.9.2**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.8.7
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (X11; CrOS x86_64 14541.0.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/110.0.0.0 Safari/537.36

LazeMSS, 2023-03-22:
Not sure 100% what the problem is

Awashcard0, 2023-03-22:
Just thought I'd put it out there