LazeMSS/OctoPrint-UICustomizer #286: [BUG] Resizable Files+Terminal Not working since version 0.1.9.4

elpeter25, 2023-03-27:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
Since version 0.1.9.4 files box cannot be resized
What is the problem?
Resize triangle icon not showing up
Can you recreate it?
Yes, have octoprint with version 0.1.9.2 and it's working (see screenshot). Updated to latest version not working
Did you try disabling plugins?
No
<hr>

**Plugins installed**
- Backup Scheduler[backupscheduler] v0.1.0
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Cancel Objects[cancelobject] v0.4.7
- Consolidated Tabs[consolidatedtabs] v1.0.1
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- GcodeEditor[GcodeEditor] v0.2.13
- HeaterTimeout[HeaterTimeout] v0.0.4
- InlineConfirm[inlineconfirm] v1.1.4
- Print Scheduler[printscheduler] v1.0.2
- PrintJobHistory[PrintJobHistory] v1.17.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.0
- Prusa Leveling Guide[PrusaLevelingGuide] v1.0.18
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Simple Filament Change Buttons[simple_filament_change_buttons] v1.1
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.2
- SmartABL[SmartABL] v0.5.1
- Terminal Messaging[terminalmessaging] v0.1.0
- TimeToFilament Plugin[TimeToFilament] v1.2.1
- Top Temp[toptemp] v0.0.2.3
- **UI Customizer[uicustomizer] v0.1.9.4**

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: false
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.8.7
- Python 3.7.5

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/111.0
[image]
[image]

LazeMSS, 2023-04-30:
Does this happen with all themes?

elpeter25, 2023-05-02:
Hi!

Thanks for your time.

I just tried a couple of themes with same behaviour, I'm using Discorded and changed for this test to Cosmo and Cyborg and even tried the Default one. All of them same behaviour, on 1.9.4 version resizing is not possible.

Cheers!

elpeter25, 2023-05-02:
Strange behaviour just noticed. If I check the "Preview" button it works well... 
I tried disabling every other plugin and behaves the same...

elpeter25, 2023-05-02:
Tried in Google Chrome, same behaviour. See attached video:

https://share.vidyard.com/watch/wNjB6Z52g2BxWxdsWeA4aK?

LazeMSS, 2023-05-08:
Could you try right clicking and inspect the file selector and select inspect - look for something like this:
[image]
notice the UICvertResize

I just can't recreate it

gkr321, 2023-05-18:
When I open octoprint I see this, with all the empty space under the files.

[image]

If I go into settings, layout turn on the preview and change the settings I get This
[image]

saving the settings doesn't help.  when I restart octoprint i'm back where I started.

LazeMSS, 2023-06-05:
Awesome thanks