LazeMSS/OctoPrint-UICustomizer #287: [BUG] Duplicate camera in side panel

dannybloe, 2023-04-12:
Howdee, 

Just installed the plugin and I noticed that the camera is duplicated in the side panel:

[image]

Looks like there a two identical divs in the DOM.

Cheers,
Danny

LazeMSS, 2023-04-12:
Would you care to upgrade to the rc of octoprint?

dannybloe, 2023-04-24:
Mm.. weird. I didn't get notified about your replies even though I have all notification settings enabled.

Anyway.. ehm... I'd rather wait till the rc is final to be honest. Isn't that soon anyway?

LazeMSS, 2023-04-30:
Strange i did not get the update either - yes i don't think it will be long before the final is out

LazeMSS, 2023-06-13:
@dannybloe was this fixed at least? :)

dannybloe, 2023-06-13:
Well, it's not working at all anymore. See https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/293

LazeMSS, 2023-08-07:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.8

Please check and reopen if not working - thank you...

closed 2023-08-07 by LazeMSS