LazeMSS/OctoPrint-UICustomizer #289: [BUG] [1.9.0] Webcam Disappears Coming Out of Fullscreen

jneilliii, 2023-04-22:
## Description
**ENTER DESCRIPTION HERE
In OctoPrint 1.9.0rc5 if you make the webcam view fullscreen, when you exit out of fullscreen mode the camera disappears. Only thing to bring it back is refreshing the page. 
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.9.4**
- Terminal Commands Extended[terminalcommandsextended] v0.1.8rc1
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.2
- WLED Connection[wled] v0.2.1
- TP-Link Smartplug[tplinksmartplug] v1.0.4rc1
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.0
- Plotly Temp Graph[plotlytempgraph] v0.1.4
- Network Health Monitor[networkhealth] v1.0.4
- Firmware Updater[firmwareupdater] v1.14.0
- Exclude Region[excluderegion] v0.3.2
- Custom Background[custombackground] v1.0.0
- Autologin Configuration[autologin_config] v0.2.1
- Active Filters Extended[active_filters_extended] v0.1.0
- OctoDash Companion[octodashcompanion] v0.0.8rc4
- Marlin EEPROM Editor[eeprom_marlin] v3.3.0
- M300 Player[M300Player] v0.1.1
- Google Drive Backup[googledrivebackup] v0.1.1
- Bed Visualizer[bedlevelvisualizer] v1.1.2rc5
- Bed Ready[bedready] v0.2.1
- Backup Scheduler[backupscheduler] v0.1.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- compressTempControls: true
- disableTermInactive: true
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: false
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.9.0rc5
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/112.0.0.0 Safari/537.36

LazeMSS, 2023-04-23:
Sighhh

jneilliii, 2023-04-23:
Tried UI Customizer on a minimally modified OctoPrint instance and the issue didn't occur, so I assume it's a plugin conflict. If I can find some time I'll try to narrow down which plugin might be causing it.

LazeMSS, 2023-04-30:
@jneilliii  based on the plugin it list you attached there not a plugin that I think should do any webcam stuff - but maybe some dom stuff -i have not tried them.

kantlivelong, 2023-05-16:
I'm seeing this as well on OctoPrint 1.9.0rc6 and UIC 0.1.9.4. Interestingly enough I also see two "Exit Fullscreen" buttons both of which yield the same result.

```
2023-05-15 21:37:17,478 - octoprint.plugin.core - INFO - 33 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Active Filters Extended (0.1.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_active_filters_extended
| !Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (1.1.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer
|  Classic Webcam (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/classicwebcam
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Custom Control Editor (0.2.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_customControl
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Event Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/eventmanager
|  Exclude Region (0.3.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_excluderegion
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.10.11) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  GCODE System Commands = /home/pi/.octoprint/plugins/gcodesystemcommands
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  OctoDash Companion (0.0.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octodashcompanion
|  Pi Support Plugin (2022.6.13) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrintTimeGenius Plugin (2.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
| !PSU Control = /home/pi/.octoprint/plugins/psucontrol
|  Slicer Thumbnails (1.0.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prusaslicerthumbnails
|  Smart Preheat = /home/pi/.octoprint/plugins/smartpreheat
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  Top Temp (0.0.2.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_toptemp
|  UI Customizer (0.1.9.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_uicustomizer
|  UPS = /home/pi/.octoprint/plugins/ups
| !Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
|  WLED Connection (0.2.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_wled
```

LazeMSS, 2023-08-07:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.8

Please check and reopen if not working - thank you...

closed 2023-08-07 by LazeMSS