LazeMSS/OctoPrint-UICustomizer #291: Issue with icons on the tab order panel:

JKinch, 2023-05-18:
Hi, while in settings i cant seem to search icons anymore while messing with TAB order/names/etc under the main tabs panel in settings. 

Get "Error searching… Error: 0 (error)". Very minor issue. Running on a a RPi3B and was installed all offline due to operating on a local only network. Since i cant find this issue elsewhere, i have a feeling the installation expects a internet connection that wasn't present. 

[image]
[octoprint-logs.zip](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/11508436/octoprint-logs.zip)

[plugin_list.json.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/11508387/plugin_list.json.txt)
S/OctoPrint-UICustomizer/assets/44818309/fd586896-9658-46aa-a409-8ae0114e425e)

Any help would be appreciated, as currently i cant fit all tabs on screen in this 4:3 monitor.

LazeMSS, 2023-05-30:
Sorry i have been away. Will look later today

LazeMSS, 2023-06-13:
Yes this bug is true - it uses the api from font awesome to search for icons. I won't fix it.

closed 2023-06-13 by LazeMSS

JKinch, 2023-06-13:
Is there not a solution for users?

LazeMSS, 2023-06-13:
I think you browser is blocking for the access - i will look a bit more

jneilliii, 2023-08-03:
You could convert to a local-only search box couldn't you? That's how my Tab Order plugin does it using [this](https://github.com/jneilliii/OctoPrint-TabOrder/blob/master/octoprint_taborder/static/js/fontawesome-iconpicker.js).