LazeMSS/OctoPrint-UICustomizer #293: WebCam Preview not working in Octoprint 1.9.0

XeroZar, 2023-06-01:
## Description
**ENTER DESCRIPTION HERE
After I updated to Octoprint 1.9.0, the webcam preview stopped working... Please fix.
<hr>

**Plugins installed**
- Active Filters Extended[active_filters_extended] v0.1.0
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Cancel Objects[cancelobject] v0.4.7
- Consolidate Temp Control[consolidate_temp_control] v0.1.10
- Fan Speed Control[fanspeedslider] v0.2.3
- Firmware Updater[firmwareupdater] v1.14.0
- Navbar Temperature Plugin[navbartemp] v0.15
- Octolapse[octolapse] v0.4.3
- OctoPod Plugin[octopod] v0.3.15
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.0
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- SpoolManager Plugin[SpoolManager] v1.7.0
- Terminal Commands Extended[terminalcommandsextended] v0.1.7
- **UI Customizer[uicustomizer] v0.1.9.4**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.9.0
- Python 3.10.6

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/113.0.0.0 Safari/537.36

jneilliii, 2023-06-01:
additional details starting here: https://discord.com/channels/704958479194128507/708230099811434497/1113595887600549948

LazeMSS, 2023-06-01:
Will look asap

jneilliii, 2023-06-08:
Link took me to the OctoPrint discord post, but I think this may be resolved with the next bug fix version of OctoPrint. @cp2004 figured out something strange with the observer being used now in the cameras relative to percentage visibility. 

Summarizing the issue, if the webcam is in the center column with tabs the stream will load for 2-3 seconds same then go black.

jjorstad2, 2023-06-08:
> Link took me to the OctoPrint discord post, but I think this may be resolved with the next bug fix version of OctoPrint. @cp2004 figured out something strange with the observer being used now in the [...]

ok thanks I was not joined on that discord server and the link did not say what server to join so I had no clue

cp2004, 2023-06-08:
> Link took me to the OctoPrint discord post, but I think this may be resolved with the next bug fix version of OctoPrint. @cp2004 figured out something strange with the observer being used now in the [...]

I can't reproduce this issue in my install, so I can't say that the change I was looking into will make a difference or not.

For me, the webcam stuff in UI Customizer and OctoPrint 1.9.0 seems to work perfectly, I can't break it - regardless of the changes I proposed to the intersection observer. Works with both my plugin and the classic webcam one as well.

dannybloe, 2023-06-12:
I got the same problem. Cam works fine but not in the side panel. Maybe these console messages help:

[image]

LazeMSS, 2023-06-13:
> I got the same problem. Cam works fine but not in the side panel. Maybe these console messages help:
> 
> [image]

Which OctoPrint version are you running?
Does it work if you disable the  prettyGcode plugin?

LazeMSS, 2023-06-13:
Could you try and post the content of the:
packed_core.js?39bcd23a:1752

basically clicking the link in console.

LazeMSS, 2023-06-13:
does it work if you move the cam to sidebar instead of the and could you post the plugins your are running?

cp2004, 2023-06-13:
> The new way multicam plugin is not supported in my plugin

It looked to me like it was pretty much working OK, with my 'Camera Streamer Control' plugin and the multicam plugin.

dannybloe, 2023-06-14:
> The new way multicam plugin is not supported in my plugin - my plugin only support classic webcam and previous 1.9.0 way (basically the same but different ways) i will look a bit more

You can do it! I believe in you 😉

nullable-eth, 2023-06-16:
After [just] installing `UI Customizer` my Webcam view will go black randomly.  If I minimize the Webcam frame it will come back.  Not un-installing `UI Customizer` though, love it too much (good work).  This only just happened, been using the same webcam for a while and has been rock solid.  I also have Octolapse installed which is using the webcam and I have noticed that when Octolapse does a snapshot it seems to black out the other camera views, this did not happen prior to `UI Customizer` though.

[image]
[image]

LazeMSS, 2023-08-07:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.8

Please check and reopen if not working - thank you...

closed 2023-08-07 by LazeMSS