LazeMSS/OctoPrint-UICustomizer #294: Auto load gcode viewer

shiftylilbastrd, 2023-06-08:
## Description
It would be great is the gcode viewer would automatically load. Currently I have to switch to the Gcode tab so octoprint analyzes the file and loads the preview and then I'm able to switch to multicamview to be able to monitor the print. A minor inconvenience but I want to saw it used to do it automatically before the 1.9 update.

Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.9.4**
- Timelapse Purger[timelapsepurger] v0.1.4
- Cura Thumbnails[UltimakerFormatPackage] v1.0.2
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.0
- Obico for OctoPrint[obico] v2.3.9
- Navbar Temperature Plugin[navbartemp] v0.15
- MultiCamView[multicamview] v0.1.0
- MultiCam[multicam] v0.3.1
- Exclude Region[excluderegion] v0.3.2
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

**Software versions**
- OctoPrint 1.9.0
- Python 3.9.2
- OctoPi 1.0.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36

LazeMSS, 2023-06-08:
Will look into it.

LazeMSS, 2023-06-13:
My guess is something breaks because of one the multicam plugins - try disabling them and run again

shiftylilbastrd, 2023-06-13:
Disabled both and no change. Still doesn't analyze the file unless I click
on the Viewer tab.

loroff, 2023-07-19:
bonjours j ai exactement le meme probleme octoprint 1.9.1 python 3.9.2 octopi 1.0.0 alors que cela marché tres bien avecla precedente version d octoprint

7s3, 2023-07-20:
I had this working, and all of a sudden it broke.
Now I have to manually go to the "GCode Viewer" Main Tab in order for it to start rendering in the GCode Viewer module that I have in one of the 3 Layout columns.

LazeMSS, 2023-08-07:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.8

Please check and reopen if not working - thank you...

closed 2023-08-07 by LazeMSS

7s3, 2023-08-08:
> Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.8
> 
> Please check and reopen if not working - thank you...

Thanks, I think it's working correctly now in 0.1.9.8

loroff, 2023-08-10:
merci beaucoup cela refonctionne correctement

LazeMSS, 2023-08-10:
agréable à entendre - bonne soirée