LazeMSS/OctoPrint-UICustomizer #295: Octoprint Browser Resolution

clarkiej, 2023-06-15:
This is not an issue but a query:
Is it possible to change teh browser screen resaolution in UI-Customiser like yo can in Themify with teh .span8 and .container parameters : 

[image]
 depending on teh screen resolution you are viewing with?

LazeMSS, 2023-06-15:
Not sure what you want, but in the advanced section you can make custom CSS

LazeMSS, 2023-07-25:
So yes you can basically just ad this in the advanced section of UI Customizer under custom css:
```
.span8{
width: 880px
}
.container{
width: 1200px
}
```
But I would not recommend it as it might break the responsiveness of the UI

closed 2023-07-25 by LazeMSS