LazeMSS/OctoPrint-UICustomizer #296: Octoprint UI Customizer

Nitehawkjv, 2023-06-23:
## Description
After installing UI Customizer when i go to the settings and choose Plugins there is no where that i can see to add other plugins
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.9.4**
- SimplyPrint Cloud[SimplyPrint] v4.1.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.0
- PrettyGCode[prettygcode] v1.2.4
- Dashboard[dashboard] v1.19.9
- OctoLight[octolight] v0.1.3
- Octolapse[octolapse] v0.4.3
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Bed Visualizer[bedlevelvisualizer] v1.1.1

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: false
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: true

**Software versions**
- OctoPrint 1.9.0
- Python 3.9.2
- OctoPi 1.0.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36 Edg/114.0.1823.51

LazeMSS, 2023-06-23:
Not sure what you mean...

LazeMSS, 2023-07-25:
Closing - this does not make sense :)

closed 2023-07-25 by LazeMSS