LazeMSS/OctoPrint-UICustomizer #298: Webcam stream loading forever in webcam side widget, despite displaying in main area

triton-nereide, 2023-07-08:
While visible and working in the main area, the video stream keeps loading forever in the video stream side-widget. Does anybody by any chance know, what´s happening here?

second question: I modified the red night theme by putting it´s css in the "advanced" css section of the plugin and editing it. Sometimes it seems to loose this modified css and displays like red night original. when I open the plugin settings and click on preview it displays correctly again.
Why this?

triton-nereide, 2023-07-11:
Have a good one! Thanks for an awsome plugin!

triton-nereide, 2023-07-12:
As I investigated this further, my thoughts are these:
I am using the new Octopi 1.0 - Octoprint 1.9.1 beta with the "new camera streamer" plugin. Your theme expects the old MJPEG streamer, so it fails to load the new webrtc stream inside the sidebar widget. (In the main area the webrtc stream loads with no issues)
I did some tests today: I deleted the data that was stored inside the classic webcam plugin and made a restart. After the restart the classic webcam plugin was enabled ((despite I disabled it some time ago) but with no data stored for the MJPG stream), alongside the Camera Streamer Plugin.
From that moment on the webrtc stream was loaded correctly in the sidebar widget (it hat an overlay text saying "webcam not configured" (because there is no data in "classic webcam plugin)), and the stylesheet losses were gone too.
So I assume that the old "classic webcam" plugin delivers something that is missing to correctly load the webrtc stream inside the side bar widget, and with this delivery it solves the css issue too.

Conclusion: If you where so kind to change your code so that the webrtc video can load in the sidebar widget, all the issues are gone, even when disabling "classic webcam" plugin.

triton-nereide, 2023-07-24:
For me it does not matter if I change to "Control" tab or not, as soon as I disable classic webcam plugin, camera streamer shows no video in sidebar, and after restart the interface loads without the custom css. As soon as I activate classic webcam plugin again all works normal. (It is necessary to delete any configuration for classic webcam plugin).
All that remain then is an overlay to the videostream, saying "webcam not configured". (this is because any config in "classic webcam plugin needs to be deleted to make the new camera streamer work)

jneilliii, 2023-07-24:
I wonder if you could utilize `self.coresettings.webcam_webcams` observable array in your viewmodel @LazeMSS to get a handle of the stream URL no matter what plugin is being used.

LazeMSS, 2023-07-25:
I will look when im back from holliday :)

LazeMSS, 2023-07-25:
> I wonder if you could utilize `self.coresettings.webcam_webcams` observable array in your viewmodel @LazeMSS to get a handle of the stream URL no matter what plugin is being used.

Then I just need to figure out which is the active entry right now and the get all the data from that entry?

LazeMSS, 2023-08-07:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.8

Please check and reopen if not working - thank you...

closed 2023-08-07 by LazeMSS