LazeMSS/OctoPrint-UICustomizer #299: Random Upload takes over screen

zvercodebender, 2023-07-13:
## Description
At random times the screen will be filled with an overlay to allow for the `Upload locally` and `Upload to SD`.  No intervention is required for this to happen and once it starts it will keep happening.  The only way back to a usable UI for the short term is to reload the page.

* Can you recreate it?  I'm not sure how this gets started, but I can make it go away for a while buy clearing the cache.
* Did you try disabling plugins?  I'm not sure which plugins it would make sense to disable for this problem

**Plugins installed**
- Exclude Region[excluderegion] v0.3.2
- GPIO Control[gpiocontrol] v1.0.8
- Octolapse[octolapse] v0.4.3
- Octoslack[Octoslack] v2.2.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.0
- Resource Monitor[resource_monitor] v0.3.15
- **UI Customizer[uicustomizer] v0.1.9.4**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: false
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

**Software versions**
- OctoPrint 1.9.1
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (X11; Linux x86_64; rv:109.0) Gecko/20100101 Firefox/114.0
[image]

jneilliii, 2023-07-13:
I've seen this happen when highlighting text and dragging the mouse, but never seen it not go away after releasing the mouse.

zvercodebender, 2023-07-14:
The only solution I have found so far is to clear the cache and reload the page.

jneilliii, 2023-07-14:
The next release of OctoPrint has an option to disable that drag and drop feature.

LazeMSS, 2023-07-25:
@zvercodebender does it happen if you run without UI Customizer?

LazeMSS, 2023-08-07:
@zvercodebender ?