LazeMSS/OctoPrint-UICustomizer #300: Hide webcamera on Control tab - don't work

fly74, 2023-07-18:
Hi Devs,

the option "Hide webcamera on Control tab" don't hide the webcam on control tab. And with this behavior the webcam can't displayed on the overview page.

Environment is:
octoprint 1.9.1
Classic Webcamplugin disabled
Multicam Plugin 0.4.0

It would be nice to show multiple webcams at one time here :):

[image]

fly74, 2023-07-18:
Ah multiple cams at a time are already here https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/206 discussed.

LazeMSS, 2023-07-25:
There is a problem with the webcam in the new release I trying to figure out the best way to fix it

LazeMSS, 2023-08-07:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.8

Please check and reopen if not working - thank you...

closed 2023-08-07 by LazeMSS