LazeMSS/OctoPrint-UICustomizer #302: [enhancement] Customize to use multiple camera feed

esrpradeep, 2023-08-05:
Hello, I'm having an ESP32-Cam used with MultiCam and ESP32 camera plugins. It would be nice to have an add-on feature to bring multiple camera feed windows to the UI. Right now, I have to switch between the cameras or select the plugin tab to get the feed of the default camera (pi camera).

LazeMSS, 2023-08-07:
Should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.8

Please check and reopen if not working - thank you...

closed 2023-08-07 by LazeMSS

esrpradeep, 2023-08-08:
Thank you. I checked your fix it is working with Multicam plugin.

Regarding support for Multiple camera, I believe it is possible to create an option for one more window for the camera. In the additional window, one can select if they want two feeds or the same feed on both windows.

Example: For my printer has two cameras feeds one from pi camera (main/default used for time-lapse), and the other one is the esp32-cam. (I have also tried using USB web camera i.e. Logi C270 from the Pi).

[image]

Use of Multicam plugin..example

[image]

ESP32-cam feed also works in your plugin -- but with just one window.
[image]

Waiting for something like multi cam feed windows ( at least two floating windows) feature in the future!

esrpradeep, 2023-08-08:
Ok, got it. All I know is two streams with two threads to handle the events (handle it as if it is from two sources). Maybe I'm simplifying too much, but it should be possible as you said. I can't ask you more, thanks for fixing the bug and for your support.