LazeMSS/OctoPrint-UICustomizer #304: Aspectratio is not respected

fly74, 2023-08-12:
When use of multicam plugin, only the classic webcam plugin shows the aspect correctly. From multicam it is always 4:3 even if 16:9 is in multicam configured.

https://github.com/LazeMSS/OctoPrint-UICustomizer/assets/38754941/558fd994-d447-4a7f-a525-b339141805d7

LazeMSS, 2023-08-14:
Does it work then control tab? Basically is the problem isolated to only then "extra" webcam view?

fly74, 2023-08-14:
The control tab works as expected with all cams.

LazeMSS, 2023-08-15:
Thanks i will look asap

LazeMSS, 2023-08-15:
@fly74  please try and add this in custom css in advanced in UI Customizer plugin:
```
.webcam_img_container,.webcam_video_container .webcam_unrotated{
    min-height: initial !important;
}

```

fly74, 2023-08-15:
@LazeMSS Hey perfect. That works! Can you implement it static?

LazeMSS, 2023-08-16:
Yes or maybe submit a MR to multicam plugin.

LazeMSS, 2023-09-02:
Closing as this should be fixed in : https://github.com/mikedmor/OctoPrint_MultiCam/releases/tag/0.4.1

closed 2023-09-02 by LazeMSS