LazeMSS/OctoPrint-UICustomizer #306: [BUG] Fullscreen webcam mode both the maximise and minimize icon

Draknars, 2023-09-18:
**Describe the bug**
Fullscreen webcam mode both the maximise and minimize icon

**To Reproduce**
Steps to reproduce the behavior:
1. What settings are you using in UICustomizer 
2.  Float/Fullscreen webcam -> Fullscreen webcam
3. Go to 'Control' Tab and Click the "Maximise" icon to put in fullscreen
4. In fullscreen, both the minimize and maximise icon overlap
5. If you click accidently on the maximise instead, it will minimize, but you won't be able to put in fullscreen anymore
6. to "Fix it" you need to reload browser

**Expected behavior**
Would expect to only have minimize when in fullscreen

**Screenshots**
Before fullscreen
[image]
In fullscreen
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser : Chrome & Edge latest version

**Additional context**
Mimizing with the correct icon or with using ESC key will not trigger this bug, only when missclicking the maximse icon while in fullscreen.

LazeMSS, 2023-11-12:
What plugins are you running for webcam? Classic webcam or? Please share a list of enabled plugins.

LazeMSS, 2023-11-12:
Should be fixed: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.9
Please reopen if not.

closed 2023-11-12 by LazeMSS