LazeMSS/OctoPrint-UICustomizer #307: Layout Can't be changed. Everything stays on the left only in a set order.

rlebkuecher, 2023-10-14:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.9.8**
- Timelapse Purger[timelapsepurger] v0.1.4
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.3
- Preheat Button[preheat] v0.8.0
- Cura Thumbnails[UltimakerFormatPackage] v1.0.2
- PrettyGCode[prettygcode] v1.2.4
- Obico for OctoPrint[obico] v2.4.8
- Navbar Temperature Plugin[navbartemp] v0.15
- Gcode Macros[gcode_macro] v1.3.0
- Dashboard[dashboard] v1.19.10
- OctoDash Companion[octodashcompanion] v0.0.7
- Nexus AI[nexus_ai] v0.1.0
- Navbar Clock[navbarclock] v1.1.4
- Macro[macro] v0.3.2
- Filament Sensor Simplified[filamentsensorsimplified] v0.3.3
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- Consolidated Tabs[consolidatedtabs] v1.0.2
- Calibration-Companion[calibrationcompanion] v1.0.1
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f

**UI Customizer settings/info**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

- New Webcam model: true
- Clasic cam: false

**Software versions**
- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi* 0.18.0 (build 2022.01.20.112925)

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/117.0.0.0 Safari/537.36 Edg/117.0.2045.60

rlebkuecher, 2023-10-14:
After updating to the most current version, the layout can not be changed. Everything stays on the left side only in a set order that can not be changed either.

rlebkuecher, 2023-10-14:
I hit the wrong button. This issue is not closed.

LazeMSS, 2023-10-19:
Do you see any errors in you console in the browser (F12)?
Do you have a screenshot?
Can you try disabling these plugins:
- Consolidated Tabs
- PrettyGCode
- Dashboard

Just to see if fixes the problem

rlebkuecher, 2023-10-20:
I've disabled the plugins you suggested. The screen shot is after disabling. No change.
[image]

jneilliii, 2023-10-20:
Press this button to see the errors Laze was looking for.

[image]

rlebkuecher, 2023-11-03:
[jneilliii](https://github.com/jneilliii), Thank you for the assistance.

This should be what you're looking for.

[image]

jneilliii, 2023-11-03:
Ah, you're running an old 1.7.3 version of OctoPrint is why. That error is because it's expecting something specific from the 1.9.0+ versions of OctoPrint.

LazeMSS, 2023-11-12:
@jneilliii thanks - maybe i should just update the requirement to force people to update.

LazeMSS, 2023-11-12:
Should be fixed: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.9
Please reopen if not.

closed 2023-11-12 by LazeMSS

rlebkuecher, 2023-11-12:
Although this is marked as completed, I wanted to add that the newest version, 0.1.9.9 corrected all my issues without needing to update OctoPrint. Last time I upgraded OctoPrint it created nothing but problems so I didn't really want to do that upgrade.
Thank you