LazeMSS/OctoPrint-UICustomizer #308: "Sync with job progress" is always active in Gcode viewer

BobSleeuw, 2023-10-25:
## Description
**Describe your problem?**
The effect of the "Sync with job progress" option in the Gcode viewer tab is always active, even when it is unchecked. This means that you can not scroll through the layers/current layer without it automatically snapping back to the current layer progress.
[image]

**What is the problem?**
If the box for the option is unchecked, the gcode view should not sync with the job progress. 

**Can you recreate it?**
Yes, just use UICustomizer and no other plugins.

**Did you try disabling plugins?**
The issue does not exist if I use Safe mode, and also not if I disable UICustomizer while other plugins are active. If I disable all non-stock plugins except for UICustomizer, the issue is present.

<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.9.8**

**UI Customizer settings/info**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: false
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

- New Webcam model: true
- Clasic cam: #classicwebcam_container

**Software versions**
- OctoPrint 1.9.2
- Python 3.7.3
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/118.0.0.0 Safari/537.36

rgitzel, 2023-10-27:
I started having the same issue after upgrading to Octoprint 1.9.3 from 1.8.x (the latest).   Browser is Mac Chrome 118.0.5993.88

LazeMSS, 2023-10-27:
I will look asap

edwardhorsford, 2023-10-28:
I'm also having this issue. Goes away if booting in safe mode, also goes away if I specifically disable UI customizer.

meier2009, 2023-10-30:
yeah, also have this issue, but a longer time.
Think since Octopi 1.9 and higher!

I now found the solution in the Help Community, but would be nice to be able to use it without this issue! :-)

If there a way to help ...

LazeMSS, 2023-11-12:
I think the problem is because the gcode viewer is doing it - i will look into it now.

LazeMSS, 2023-11-12:
Should be fixed: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.9
Please reopen if not.

closed 2023-11-12 by LazeMSS

meier2009, 2023-11-12:
Works! 😎
Thx a lot!

BobSleeuw, 2023-11-12:
Works great, thanks for the fix! @LazeMSS