LazeMSS/OctoPrint-UICustomizer #310: bug

loroff, 2023-11-13:
**PLEASE USE THE "Report a bug" button found in settings**

That button will prefill everything and you don't have to do anything besides describe the problem.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. What settings are you using in UICustomizer
2. Go to '...'
3. Click on '....'
4. Scroll down to '....'
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

LazeMSS, 2023-11-13:
Please write in English.

But basically its a compromise between performance and functionality. So for now its not easy fixable.

loroff, 2023-11-15:
ok thank you for your reply and for your work

closed 2023-11-16 by LazeMSS