LazeMSS/OctoPrint-UICustomizer #312: [HELP] How to change color of .btn-navbar to make it better suited for a dark theme?

redjacketpress, 2023-11-23:
I've tried tinkering with the CSS, but the nature of that button makes it very difficult to customize. I've even tried hiding it by default if the display width is too small (I would reveal it as needed), but that doesn't seem to be working, either.

Any advice would be appreciated!

LazeMSS, 2023-11-24:
Ahh we might have missed that in the themes but something like this might work:

#navbar a.btn.btn-navbar {
    background-image: none !important;
    background-color: #888;
}

#navbar .btn-navbar .icon-bar {
    background-color: #aaa;
}

redjacketpress, 2023-11-24:
That was embarassingly simple -- thanks!

I tried changing those kinds of values in Google Chrome (I also tried commenting out all of the background-image calls for the various browsers), but that never seemed to work -- I was never able to get to the button itself.

closed 2023-11-27 by LazeMSS