LazeMSS/OctoPrint-UICustomizer #313: Left column

TMTYD, 2023-11-29:
## Description
**ENTER DESCRIPTION HERE
When I create multiple folders in the file column, it looks like this is pushed together instead of being spread across the height of the screen
[image]

<hr>

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Calibration Tools[CalibrationTools] v1.0.4
- Cost Estimation[costestimation] v3.5.0
- Dashboard[dashboard] v1.19.10
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.28.0
- FileManager[filemanager] v0.1.6
- Firmware Updater[firmwareupdater] v1.14.0
- Marlin GCode Documentation[marlingcodedocumentation] v0.18.0
- OctoApp[octoapp] v1.3.3
- Preheat Button[preheat] v0.8.0
- PrintJobHistory[PrintJobHistory] v1.17.1
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.1
- PSU Control[psucontrol] v1.0.6
- PSU Control - Fritz!Box[psucontrol_fritz] v1.0.3
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.3
- SpoolManager Plugin[SpoolManager] v1.7.0
- TimeToFilament Plugin[TimeToFilament] v1.2.1
- **UI Customizer[uicustomizer] v0.1.9.9**

**UI Customizer settings/info**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

- New Webcam model: true
- Clasic cam: 

**Software versions**
- OctoPrint 1.9.3
- Python 3.9.2

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:120.0) Gecko/20100101 Firefox/120.0

LazeMSS, 2023-12-05:
" being spread across the height of the screen"

What do you mean?

TMTYD, 2023-12-08:
Ich meine das so, wenn ich die anderen menüs einklappe, wie z.b. verbindung überhalb des Menüs Dateien, dann bleibt es auf der selben größe, wie wenn ich das obere Menü ausgeklappt lasse.

LazeMSS, 2023-12-11:
I will look.

closed 2025-04-08 by TMTYD