LazeMSS/OctoPrint-UICustomizer #314: Octorpint top bar and bottom bar obscures Pretty GCode fullscreen view

hyperair, 2023-12-23:
## Description

When entering Pretty GCode fullscreen view, the top and bottom bar still float above everything, which obscures the unfullscreen button among other things.

I have managed to workaround it with this custom CSS:
```css
body:has(.pgfullscreen) #navbar {
    z-index: -1;
}
```

Things look like this now in Pretty GCode's fullscreen mode:
[image]

<hr>

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- M73 ETA Override[m73etaoverride] v1.0.4
- OctoKlipper[klipper] v0.3.9.5
- Dashboard[dashboard] v1.19.10
- PrettyGCode[prettygcode] v1.2.4
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.3
- **UI Customizer[uicustomizer] v0.1.9.9**

**UI Customizer settings/info**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: false
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

- New Webcam model: true
- Clasic cam: #classicwebcam_container

**Software versions**
- OctoPrint 1.9.3
- Python 3.11.2

**Browser**
- Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0.0 Safari/537.36

LazeMSS, 2023-12-27:
Thanks - will include the fix asap

LazeMSS, 2024-02-12:
FIxed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.91 - reopen if not fixed

closed 2024-02-12 by LazeMSS