LazeMSS/OctoPrint-UICustomizer #315: Layout appears to have a maximum of 11 not 12, and camera is in Letterbox

mrbchambers, 2024-02-02:
## Description
Just recently updated octoprint to OctoPrint 1.10.0rc1, and since then; when I try to set the layout to a total width of 12, the 3rd column appears under the first column. Also webcams appear in letterbox (black area above and below the video feed), though this only appears in UI Customiser, in all other plugins the video feed is as expected (no black area around the video), and strangely when using dashboard in UI Customizer, the video is fine in the Dashboard panel, but not in the other UI Customiser Webcam containers (I don’t know the right terminology, though hope this makes sense).

I don’t know what is the underlying cause of this problem, though it wasn’t there before the update, but in fairness I have posted on the octoprint forum about this problem too.

I have added 2 pictures: Camera views to show the webcam effect, and 3 4 5; to show what happens when I set the columns to 3;4;5, the camera view photo was taken with a column layout of 3;4;4.

This is totally repeatable, and happens on both of my identical Raspi systems (Raspi 4, with raspicam module 3 noir)

Did you try disabling plugins? Yes, too many combinations to write down, though the results were the same.

Did you remember to update the subject?** All updates carried out, every plugin is the latest version.
<hr>

**Plugins installed**
- BGCode[bgcode] v0.2.0
- Dashboard[dashboard] v1.19.10
- Firmware Updater[firmwareupdater] v1.14.0
- Octolapse[octolapse] v0.4.5
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.1
- Simple Emergency Stop[simpleemergencystop] v1.0.5
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.6
- Tempsgraph Plugin[tempsgraph] v0.3.7
- **UI Customizer[uicustomizer] v0.1.9.9**

**UI Customizer settings/info**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

- New Webcam model: true
- Clasic cam: 
[image]

**Software versions**
- OctoPrint 1.10.0rc1
- Python 3.9.2
- OctoPi* 1.0.0cam (build 2023.
[image]
07.20.144556)

mike-bee, 2024-02-05:
Same here! Camera letterboxed and right column items displayed in middle column at 3:6:3 layout.

mike-bee, 2024-02-12:
No flipping or rotation in my case.

LazeMSS, 2024-02-12:
Fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.9.91- please check and reopen if not fixed

closed 2024-02-12 by LazeMSS

mike-bee, 2024-02-12:
Working again, here as well. Thanks!

mrbchambers, 2024-02-12:
Brilliant, all woking on the latest RC.

Thank you

Hope you enjoyed your holiday.