LazeMSS/OctoPrint-UICustomizer #318: Webcam not placed in right column, even if set to

fly74, 2024-02-11:
In OctoPrint 1.10.0rc1 - Python 3.11.5 the webcam is showing at bottom uf the middle column. May it is fixable before release of 1.10.

[image]

LazeMSS, 2024-02-11:
I will look asap

LazeMSS, 2024-02-12:
Could you try disabling the dashboard plugin?

fly74, 2024-02-12:
> Could you try disabling the dashboard plugin?

Same with Dashboard disabled.
[image]

LazeMSS, 2024-02-12:
How do you expect it to behave and also please share also your webcam settings

foosel, 2024-02-12:
Pretty sure this is a duplicate of #315.

fly74, 2024-02-12:
@foosel You're right.

closed 2024-02-12 by fly74