LazeMSS/OctoPrint-UICustomizer #320: Webcam not activated

fluky, 2024-02-17:
## Description
**ENTER DESCRIPTION HERE
Describe your problem? Webcam area stuck on "Web stream loaded".  Can access webcam fine at webcam/?action=stream url.  Works on Chrome, fails on Safari.
Can you recreate it? Yes
Did you try disabling plugins? It worked before installing PolarCloud but disabling doesn't fix it.
<hr>

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Camera Settings[camerasettings] v0.4.3
- Creality Temperature[CrealityTemperature] v1.2.4
- Creality-2x-temperature-reporting-fix[ender3v2tempfix] v0.0.4
- M73 Progress Plugin[m73progress] v0.2.1
- PolarCloud[polarcloud] v1.18
- **UI Customizer[uicustomizer] v0.1.9.91**

**UI Customizer settings/info**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: true
- themeLocal: false

- New Webcam model: true
- Clasic cam: #classicwebcam_container

**Software versions**
- OctoPrint 1.9.3
- Python 3.9.2
- OctoPi* 1.0.0 (build 2023.10.09.151442)

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.2.1 Safari/605.1.15

LazeMSS, 2024-02-17:
tried uninstalling PolarCloud ?

fluky, 2024-02-18:
I have, no difference.  Though the fact that it's only an issue on Safari makes me thing it's more of an issue with the the front end than the backend?

fluky, 2024-02-18:
On another note, after uninstalling PolarCloud, no difference, but after then installing Obico the camera is back in Safari.  Possibly something with cache but it appears the combo of Safari, PolarCloud, and UI Customizer has something going on.

LazeMSS, 2024-02-18:
UI Customizer changes very little on how the webcam works - it basically just moves the content arround.

Injineers, 2024-04-11:
i get /have the same issue. 

webcam will load for a little while, but then get teh webcam not loaded error. however every time i go directly to the stream its working.

If I remove the UI customizer, I no longer have this issue (but like the UI customizer , so dont mind when I want to see the cam to click the stream link if the embeded one dies

LazeMSS, 2024-04-12:
@Injineers what webcam module are you using and streaming type (mjpeg?)