LazeMSS/OctoPrint-UICustomizer #322: Octoprint gets stuck on "Still Starting Up..." if UI Customizer is enabled.

UnworthySeraph, 2024-05-10:
Octoprint Version 1.10.0
Running on Raspberry Pi 4B 4GB RAM

Plugins installed: 

  | Arc-Welder 
  | Automatic Shutdown (1.0.0+u.bb71e8f)
  | Camera Settings (0.4.3) 
  | Cancel Objects (0.5.0) 
  | Consolidate Temp Control (0.1.10) 
  | Dashboard (1.19.11) 
  | Exclude Region (0.3.2)
  | FileManager (0.1.6) 
  | HeaterTimeout (0.0.4) 
  | Obico for OctoPrint (2.4.13) 
  | Octolapse (0.4.5) 
  | Power Failure Recovery (1.2.1) 
  | PrettyGCode (1.2.4) 
  | PrintTimeGenius Plugin (2.3.3) 
  | PSU Control (1.0.6) 
  | Resource Monitor (0.3.16) 
  | Slicer Thumbnails (1.0.7) 

Problem:
When starting the Octoprint server, it will get stuck on "Still Starting Up..." I have to power cycle the pi to force it to boot into safe mode. Disabling the UI Customizer Plugin was the only thing that worked to get it to boot normally. This issue potentially appeared after I updated to Octoprint Version 1.10.0. 

[octoprint.log](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/15268810/octoprint.log)

LazeMSS, 2024-05-11:
Strange - i will look now

LazeMSS, 2024-05-11:
My PSU control wizard seems to be stuck - it was shown after the upgrade telling me standard gpio was no longer supported but i was unable to finis the wizard. But a reload fixed it.

But could you try disabling Obico, PrettyGCode, Dashboard plugins and trying to see if that fixes it - and if so try enabling them one after one with a reload of the UI after each enable of them.

UnworthySeraph, 2024-05-15:
So I disabled Obico, PrettyGCode, and Dashboard and that did the trick. Octoprint was able to start without issues with UI Customizer enabled. I then re-enabled the plugins one by one and it's still working fine. However, in between disabling those 3 plugins and re-enabling them, I updated Octoprint to 1.10.1. Perhaps that had something to do with it? Thanks for your help!

LazeMSS, 2024-05-16:
I will check - but I don't think 10.0.1 has any changes. But Dashboard has been know to be troublesome. I will close - reopen if this happens again.

closed 2024-05-16 by LazeMSS