LazeMSS/OctoPrint-UICustomizer #323: Unable to view, load and install themes

3D-enigma007, 2024-05-26:
## Description
**ENTER DESCRIPTION HERE
Describe your problem? 
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

Unable to view, load and install themes
[image]

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Cancel Objects[cancelobject] v0.5.0
- Creality-2x-temperature-reporting-fix[ender3v2tempfix] v0.0.4
- Cura Thumbnails[UltimakerFormatPackage] v1.0.2
- Dashboard[dashboard] v1.19.11
- FileManager[filemanager] v0.1.6
- FreeMobile Notifier[freemobilenotifier] v0.1.6
- Jog Wheel Control[jogwheel] v0.0.2
- Marlin EEPROM Editor[eeprom_marlin] v3.3.0
- MarlinSlider Plugin[marlinslider] v0.3.1
- Obico for OctoPrint[obico] v2.4.13
- OctoDash Companion[octodashcompanion] v0.0.7
- OctoEverywhere![octoeverywhere] v3.3.4
- Octolapse[octolapse] v0.4.5
- Octoprint Display ETA[display_eta] v2.2.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.3
- Sdwire[sdwire] v0.1.5
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.7
- Telegram Notifications[telegram] v1.6.7
- Timelapse+[timelapseplus] v1.4.1
- **UI Customizer[uicustomizer] v0.1.9.91**
- ZupFe For Octoprint[zupfe] v0.1.3

**UI Customizer settings/info**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

- New Webcam model: true
- Clasic cam: #classicwebcam_container

**Software versions**
- OctoPrint 1.10.1
- Pitone 3.9.2
- OctoPi* 1.0.0 (build 2023.03.02.160739)

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/125.0.0.0 Safari/537.36

LazeMSS, 2024-05-27:
Your RPI/Octoprint or localpc installation can't connect to the internet - the themese are downloaded of the internet.

LazeMSS, 2024-05-28:
Maybe the reverse proxy stuff - the themes are loaded in your browser so that might that - look in the browsers developer console mayb?