LazeMSS/OctoPrint-UICustomizer #325: Dark Green, Discoranged, Discorded and Matrix themes are all bright and the same as the Default Theme

imerkow, 2024-07-16:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.1.1
- Camera Settings[camerasettings] v0.4.3
- Cancel Objects[cancelobject] v0.5.0
- Dashboard[dashboard] v1.19.11
- Exclude Region[excluderegion] v0.3.2
- Firmware Updater[firmwareupdater] v1.14.1
- HeaterTimeout[HeaterTimeout] v0.0.4
- Navbar Temperature Plugin[navbartemp] v0.15
- Preheat Button[preheat] v0.8.0
- PrettyGCode[prettygcode] v1.2.4
- PrintTimeGenius Plugin[PrintTimeGenius] v2.3.3
- Resource Monitor[resource_monitor] v0.3.16
- Simple Emergency Stop[simpleemergencystop] v1.0.6
- Slicer Thumbnails[prusaslicerthumbnails] v1.0.7
- Title Status[title_status] v0.0.5
- **UI Customizer[uicustomizer] v0.1.9.91**

**UI Customizer settings/info**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- filesFullHeight: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true
- saveAccordions: false
- themeLocal: false

- New Webcam model: true
- Clasic cam: #classicwebcam_container

**Software versions**
- OctoPrint 1.10.2
- Python 3.9.2
- OctoPi* 1.0.0cam (build 2024.06.18.101629)

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/126.0.0.0 Safari/537.36 Edg/126.0.0.0

LazeMSS, 2024-07-21:
Strange never seen that before - tried different browsers?

imerkow, 2024-07-22:
No, not yet. Thanks for the heads up! I will check once our Pi is back online. By the way, I used Edge for testing. Perhaps it is finally time to switch. Microsoft already started to annoy me with their Copilot.

imerkow, 2024-08-16:
We finally installed our Pi again. I tried it now with Firefox. Still the same issue. All the other themes do work except the ones mentioned in the title.
[image]
[image]

LazeMSS, 2024-08-18:
Can you try and open the developer tools in firefox and look in the console for any errors and share them?

imerkow, 2024-08-19:
The resource from “http://octoxl.local/plugin/uicustomizer/theme/discoranged.css?v=0.2.9.8” was blocked due to MIME type (“text/html”) mismatch (X-Content-Type-Options: nosniff).

LazeMSS, 2024-08-20:
It seems like the css files have not been downloaded when selecting it - maybe your pi can not connect to the internet?

imerkow, 2024-08-21:
How to check that? Perhaps our firewall is blocking the download. Although the download of most plugins works. Not sure, if some plugins were already abandoned. Also, other themes do work. Just not those 4 in the title.

imerkow, 2024-08-21:
Is this a sign of that the download was blocked by our firewall?
[image]

LazeMSS, 2024-08-21:
Yes this is 100% a download problem - the reason why some themes works is because they are included in the actual plugin.
The included ones are here:
https://github.com/LazeMSS/OctoPrint-UICustomizer/tree/main/octoprint_uicustomizer/static/themes/css

and the "downloadable" ones are here:
https://github.com/LazeMSS/OctoPrint-UICustomizerThemes/tree/main/css

my gues is that anyone on the last list and not on the first list is those not working.

The RPI needs to be able to download software.

closed 2024-08-22 by LazeMSS