LazeMSS/OctoPrint-UICustomizer #326: More a question...

wrenhal, 2024-11-21:
Does the UI Customizer save the layout per device via cookies, or is it saved for all devices?

LazeMSS, 2024-11-21:
The layout - do you mean what is shown where - ie the webcam on the right or left column etc.

Its stored in the Octoprint preferences so its global/all devices.

wrenhal, 2024-11-24:
Sorry, I meant layout in general.  Chosen theme, etc...  I was wondering if the theme selection could be per device viewing the page.  I have a laptop I use to monitor Octoprint from , but I also have a Dell Tablet I was going to setup next to the printer and I wanted it to have a different theme/ui setup than when I'm on the laptop.  I hope that makes sense.

LazeMSS, 2024-11-25:
Thats not possible at the moment - the settings are stored as part of Octoprint settings - it would be possible to make it save to the local browser as a kind of overwrite.

wrenhal, 2024-11-25:
OK.  Thanks